SterlingPeet/fprime-arduino #15: Building math component for fprime-arduino platforms

capsulecorplab, 2023-07-25:
Shot in the dark, but attempted to build the [fprime-tutorial-math-component](https://github.com/fprime-community/fprime-tutorial-math-component) project using @ethancheez's [fprime-arduino](https://github.com/ethancheez/fprime-arduino/tree/rpipicow) fork on the `rpipicow` branch with the `config` folder and `settings.ini` (with `default_toolchain` set to `rpipicow`) from [fprime-arduino-led-blinker](https://github.com/fprime-community/fprime-arduino-led-blinker), but encountered a few cmake errors
[image]
Has anyone else had more success attempting to build the math component for an `fprime-arduino` platform?

ethancheez, 2023-07-25:
Currently the math component tutorial does not support baremetal because it uses Linux drivers.

capsulecorplab, 2023-07-25:
Ah, that makes sense

ethancheez, 2023-07-25:
If you want, you could create an Arduino version of the math component tutorial. The `Components/` directory would remain untouched. Then you can use my Arduino cookiecutter to create an Arduino-based `MathDeployment` by following [this](https://github.com/fprime-community/fprime-arduino-led-blinker/blob/main/docs/initial-integration.md#creating-the-ledblinker-deployment) from my `fprime-arduino-led-blinker` tutorial. Then, add the Math components to your topology.

capsulecorplab, 2023-07-25:
This might be cheating, but I just forked the `fprime-arduino-led-blinker` and added the math components (with deployment + updated topology) [here](https://github.com/capsulecorplab/fprime-arduino-led-blinker/tree/arduino-math-deployment-teensy41) :upside_down_face: So far, the build seems to be working~

[image]

Will have to try to build/port/test the deployment for the rpipicow tomorrow morning...

capsulecorplab, 2023-07-25:
Not much luck with trying to build the math component for the rpipicow :confused: 

[image]
[image]
[image]
[image]
[image]

ethancheez, 2023-07-25:
I can give a few pointers right now (I was able to deploy math component on my Teensy), but I will be able to help more in detail later today.

After taking a peek at your [fork](https://github.com/capsulecorplab/fprime-arduino-led-blinker/tree/arduino-math-deployment-teensy41), the `MathDeployment/` directory you have is still built for Linux systems. Instead, you can directly add the math instances to the `LedBlinker` deployment. 

You also need to edit the `config/` files as those configurations were specific to the `fprime-arduino-led-blinker` project. Or, you can use the default config inside of `fprime/config/` if you are using a board with sufficient memory (i.e. Teensy 4.1).

capsulecorplab, 2023-07-26:
Still waiting on my Teensy 4.1 to arrive in the mail, so just testing with the rpipicow for now~

What should I look out for as far as editing the config files?

ethancheez, 2023-07-26:
> Still waiting on my Teensy 4.1 to arrive in the mail, so just testing with the rpipicow for now~
> 
> What should I look out for as far as editing the config files?

The main ones are the telemetry configurations (`TlmChanImplCfg.hpp`) and command dispatcher (`CommandDispatcherImplCfg.hpp`). The macros defined in here depend on how many components/commands/telemetries you have.

ethancheez, 2023-07-26:
Feel free to take a look at [this](https://github.com/ethancheez/fprime-arduino-math-component)